MaretEngineering/MROV #13: Rewrite the code that parses the string from serial and place it in a separate method

Eclip5e5, 2015-11-30:


Eclip5e5, 2015-11-30:
Something like this:
```C 
void parseSerial() {
  int motorDelim = (int)'/';

  int index = 0;
  String input = "";

  if (Serial.available() > 0) {
    if (Serial.read() != (int)'!') { return; } //Don't parse wrong
  }

  while (Serial.available() > 0 && index < NUM_MOTORS) {
    int numeral = Serial.read(); //Read in an individual number
    if (numeral = motorDelim) {
      int motorInputLength = input.length() + 1; //+1 to make room for the null terminator
      char motorInputArr[motorInputLength]; //convert string
      input.toCharArray(motorInputArr, motorInputLength);
      motorInputPower[index] = atoi(motorInputArr);

      index++; //Next motor
      input = ""; //Reset value
    }
    else {
      input += (char) numeral; //Adds the individual numbers together
    }
  }
}
```

notzane, 2016-01-17:
@Eclip5e5 reopen this if we still need it in the new buffer version

closed 2016-01-17 by notzane